sparkfun/SparkFun_DataLogger #2: Do you plan to upload source for system?

markwjkelly, 2023-07-19:
We are exploring using the Datalogger_IoT_9DoF. We deploy our systems in secured locations and need to be able to present our source code as part of the project to get clearance.  Do you plan to release the code for this system, or do we need to create our software solution?

gigapod, 2023-07-19:
Moved this issue to our firmware repo - which is separate from the hardware repos that use this firmware

falkorichter, 2024-01-07:
At least the label is not on the website anymore.
[image]

falkorichter, 2024-01-09:
[image]