sparkfun/SparkFun_DataLogger #3: Using Flash Download Tool of Espressif to update the firmware.

Vanbk, 2023-08-03:
Hello. 
I accidentally download my own firmware to the board. Therefore, the firmware update instruction provided by SparkFun doesn't work. Could you please support me to recover my board? I need the compiled binary files that can be used with the  Espressif Flash Download Tool .  

Thank you very much!

gigapod, 2023-08-03:
Hi -

We just published version 01.00.01 of the firmware - the version the 9DOF board ships with.  This includes the needed files and esptool.py command line used to recover the board. 

If you have issues, let us know (here)! 

Thanks!

-Kirk

closed 2023-08-03 by gigapod

mvschaefer, 2024-03-25:
@gigapod - where can i find the esptool.py and instructions for re-flashing a .bin file?

gigapod, 2024-03-26:
The command needed are in the notes of this release: https://github.com/sparkfun/SparkFun_DataLogger/releases/tag/v01.00.01

esptool.py can be installed using pip: https://docs.espressif.com/projects/esptool/en/latest/esp32/installation.html

Once the board is reflashed, you can use the built in update tools to upgrade to the latest firmware version.