sparkfun/SparkFun_DataLogger #4: Are the firmware sources available?

gmacario, 2023-09-17:
Hello,

I just discovered this interesting product and I would like to understand if

1. The source code is available. Unless I missed it, I could only find the binaries in this repository
2. If so, under which licenses are the sources released

Thanks in advance!

gigapod, 2023-09-19:
Hi @gmacario -

The source code for our DataLogger IoT products is not open source.  

If you have questions on functionality, or have any further requirements to evaluate the product, do let us know. 

Regards,
-Kirk

closed 2023-09-19 by gigapod

gmacario, 2023-09-19:
> Hi @gmacario -
> 
> The source code for our DataLogger IoT products is not open source.
> 
> If you have questions on functionality, or have any further requirements to evaluate the product, do let us know.
> 
> Regards, -Kirk

Thanks @gigapod for the clear answer. I had hoped to receive a different answer, though, but it's your choice.

At any rate if you are open to discuss a possible commercial licensing of the code I will be glad to discuss it further.
If so, please get in touch with me at https://www.linkedin.com/in/gmacario/

jamiebegin, 2024-11-10:
@gmacario So you're saying that you're not using any GPL libraries in this product's code?

gmacario, 2024-11-10:
> @gmacario So you're saying that you're not using any GPL libraries in this product's code?

@jamiebegin maybe you wanted to ask this direct question to @gigapod instead?

gigapod, 2024-11-10:
Hi @gmacario , @jamiebegin:

Yep - a direct question to me is the way to go here.

As for GPL v3 - not to our knowledge as a result the reviews performed.  I'd schedule another review just to double check but ...  

The good news - We're about to open source this and the underlying framework.   We just need to cleanup some of the documentation we have.  We were looking at later this year, but given this thread, I'll see what it will take to just open it up quietly as a "early preview/alpha" this week.  With some documentation lacking, it might be *confusing*, but no harm in doing it really. 

FWIW - The rough doxygen docs of the underlying framework  - "flux" - are actually available here [flux-sdk](https://docs.sparkfun.com/flux-sdk/). Some of this is outdated and needs updating, but it gives you an idea of the underlying scope of the project.

gigapod, 2024-11-14:
Hi @gmacario , @jamiebegin:

You can be the first to know - this firmware is now open and public. 

The SparkFun DataLogger IoT firmware is located [here](https://github.com/sparkfun/sfe-datalogger)

The flux-sdk, which the DataLogger uses is located [here](https://github.com/sparkfun/flux-sdk)

We'll be working to note this in this repo shortly ....

gmacario, 2024-11-15:
That's an excellent news, thank you very much @gigapod!