sparkfun/SparkFun_DataLogger #5: Possibility of Releasing the Firmware as a Dynamic Library

atabou, 2023-10-01:
Hello,

I just found out that the firmware is closed-source.

Would it be possible to provide it as a dynamic library with a corresponding interface contained in a header file?

That way we would be able to interact with the firmware and develop on top of it. For example, you could provide us with a function to add a function pointer to the list of procedures that need to run when logging.

An example code could be:

```
string readVoltage() {...}

firmware.log(&readVoltage);
firmware.log(&addStuff);
firmware.start();
```

I really think this would improve the usability and extensability of the product while not divulging the information you want to keep closed-source.

Thank you!

gigapod, 2023-10-27:
HI @atabou,

We would love to get to the point to support this - its a goal to provide an SDK/external access to the firmware framework we've developed.  But we're pretty far away from delivering this solution - or understanding all the technical requirements.

If you have any particular sensor, or other more definitive functional set needed, we can implement that fairly quickly.

For what it's worth - some extensibility uses we see near term include
- Log values from a GPIO pin
- Log value sent to the datalogger via an UART connection
- Log values read from a BLE device

Not the immediate answer you're looking for (SDK), but if you have any other ideas please let us know.

Regards,
-Kirk

edubu, 2023-12-23:
Hello All,

I also would like to add some change myself but the firmware is closed-source. The BME68x with Qwik connect is automatically identified by the datalogger and is super convienient to use. Although, it only returns the humidity, gas resistance, and pressure. It is commonly used along with Bosch Proprietary Software to measure the (IAQ)Indoor Air Quality. To do this, I would need access to the source code to add the [https://github.com/boschsensortec/BSEC-Arduino-library](url) to the code. Otherwise, there is no way to retrieve the IAQ using this board and sensor together which removes half the capabilities of the sensor.

Is there a way you can add this or give me instructions on how to implement this? I would be interested in contributing to add this functionality. Would probably be better as an optional add-on since bsec seems to consume RAM and ROM quite heartily.

Thanks,
Elliot

gigapod, 2024-01-03:
We'll take a look at this for a future release - the implementation / integration of this library into the framework looks straightforward enough. As noted, the ROM/RAM could be a concern, but we need to give it a try first to determine a path forward.  But we'll take a look.

daz10000, 2024-05-11:
I understand you wanting to keep it closed.  It might be worth nothing that somewhere in the docs at top - I spent some time trying to help someone troubleshooting #32 and spent 30 mins looking for the source to see if I could understand why it was happening.

gigapod, 2026-01-21:
This is now open source, and can be used as desired.

The core is the [flux-sd](https://github.com/sparkfun/flux-sdk) which supports ESP32 and RP2 platforms (mostly)

The datalogger implementation is located [here](https://github.com/sparkfun/sfe-datalogger)

closed 2026-01-21 by gigapod