sparkfun/SparkFun_DataLogger #6: Issue with TIME - jumping by four hours

PaulZC, 2023-10-19:
Based on this customer comment:

https://www.sparkfun.com/products/20594#comment-65305d4e8e19f166d1429ed5

"
I'm having the worst time with TIME. I'm logging the epoch time on the data logger and the time from the RTC (8803) at 1Hz. It seems every hour (3600 samples) the time jumps by 4 hours. It starts at the correct time. Both the RTC time and the epoch time do the same jump at the same sample. Here is a plot:

[image]
"

PaulZC, 2023-10-19:
I've asked the customer which time zone they are using. A jump of exactly four hours suggests this could be a time zone issue? But is the jump happening in the time firmware, or the RTC 8803 driver?

gigapod, 2023-10-19:
I believe the issue might be related to how the clocks are updated in the DataLogger, at least part of the behavior seen.  You can set a reference clock and an update interval. The default interval is 60 minutes or 3600 secs. 

So every 3600 secs, the epoch is grabbed from the ref clock (the RTC in this case) and set in the current clock (the system clock). 

The set of the local time just calls settimeofday() with a pointer to empty (UTC) timezone struct {0,0}, but it's possible that this value should just be NULL.  

Also, a new time setting API should be used clock_settime()? 

Looking into this...

PaulZC, 2023-10-19:
Just a thought. Could this be a quarter-hour thing? The RV8803 library and Flux driver support Time Zones in increments of 15 minutes. Could an increment of 4 * 15mins = 1 hour accidentally becoming an increment of 4 hours?

gigapod, 2023-10-19:
Hmm - not sure. It's either the wrong secs from the RTC - offset by a timezone (which doesn't look like it is ) is being used when by settimeofday() call to esp32....

One thing I noticed, the DL is using  a 1970 based epoch, but the calls to the RTC are not setting that flag on get/set epoch. 

Probably not an issue when just making get/set epoch calls - secs are secs, but issue if getting formatted time values.

PaulZC, 2023-10-19:
If I set ```Offset Epoch``` back to false, and reset the RV8803 Epoch, all is well again...

gigapod, 2023-10-27:
Just for an update on this -- we are investigating and re-evaluating how our RTC/clock system was implemented. 

Our goal is to have a fix for this issue in the mid-November firmware update.

gigapod, 2023-10-30:
I've found the issue - it's a bug in our RV8803 Arduino Library - [RV8804 Library](https://github.com/sparkfun/SparkFun_RV-8803_Arduino_Library).

The getEpoch() call is returning a value for the local epoch value, not UTC epoch. It actually calls getLocalEpoch() to calculate the value - which uses mktime() that actually adjust the timezone. [This](https://github.com/sparkfun/SparkFun_RV-8803_Arduino_Library/blob/3919185a67bf8f88423baf423a9b9922dbcbd370/src/SparkFun_RV8803.cpp#L468) is the issue.

Will work on getting our RV8804 library fixed, and roll that in our next firmware release v1.1 in a couple weeks. 

-K

scootergarrett, 2023-11-18:
Thanks to everyone who worked on this. I made the post original post on SparkFun. I was a little frustrated so I stopped thinking about this until I saw an update available. I gave it a try last night and all seem good with time skips.
I do have another question regarding time and the RTC 8803. I have the correct Epoch time set on it but can't seem to set the time zone on the RTC. When it displays the time it is GMT with no time zone +-. One of the output options is ISO8601 format with time zone, but it just always at 0. Any thoughts on setting the time zone on the RTC?  My quick research makes me think the RTC module doesn't even do time zones.
Maybe this is not the correct place to ask. 
Thanks for the other fix, and I get it time is tough to get right.
[image]

PaulZC, 2023-11-20:
Hi @scootergarrett ,

After you reported this issue, we realized that using the RV8803 library time zone feature was not the right thing to do.

Correct, the RV8803 does not support time zones directly. But it has one spare byte of battery-backed RAM where we can store the time zone as int8_t in 15-minute increments. Our Arduino Library writes the time zone to that byte and then uses it when (e.g.) assembling the ISO-TZ-formatted time.

But, we realized that the time zone needs to be managed centrally, so that other time sources (like GNSS, which only does UTC) can be matched to NTP via WiFi. It also makes daylight saving easier to manage. The issue with the time jumps was caused by the time zone being doubly-applied from the RV8803 and the ESP32 mktime function.

In firmware 1.1.1, you will find the time zone setting in the Time Setup menu:

[image]

The Logger menu lets you set the timestamp mode:

[image]

Please see this section of the manual for more detail: https://docs.sparkfun.com/SparkFun_DataLogger/configuration/#general-time-sources

There is a full list of POSIX time zones here: https://github.com/nayarsystems/posix_tz_db/blob/master/zones.csv

Best wishes,
Paul

scootergarrett, 2023-11-24:
Thanks again. Time seems so simple to us humans but we sure have made it a trick for computer (time zones, daylights saving ... some places, leap years, leap seconds)