sparkfun/SparkFun_DataLogger #7: FEATURE REQUEST - ADS1015 Unsigned Mode

bherbruck, 2023-10-27:
The "DataLogger IoT" currently faces an issue related to the ADS1015 analog sensor, impacting the accuracy and usability of analog data acquisition. This feature request aims to resolve the problem by introducing an "unsigned mode" for the ADS1015 sensor.

1. **Current Behavior**: The ADS1015 sensor, as integrated into the "DataLogger IoT" project, exhibits a behavior where it outputs signed values ranging from -2047 to 2047. This signed representation poses challenges when collecting analog data, as it limits the effective utilization of the sensor's full 12-bit range.

2. **Impact on DataLogger IoT**: The existing behavior of the ADS1015 sensor negatively impacts the accuracy of analog data collected by the "DataLogger IoT" project. This deviation from expected behavior leads to data inaccuracies, which can undermine the reliability of the data logging system.

3. **Request**: We propose the implementation of an "unsigned mode" specifically within the "DataLogger IoT" project for the ADS1015 sensor. In this mode, the sensor would output values in an unsigned format, utilizing the entire 12-bit range from 0 to 4095. This change will enhance the compatibility and accuracy of the ADS1015 sensor within the context of the "DataLogger IoT."

4. **Comparison to OpenLog Artemis**: It is worth noting that the OpenLog Artemis, a related device, operates with an unsigned output format, which simplifies data acquisition and ensures consistency. Aligning the ADS1015 behavior within the "DataLogger IoT" project with the OpenLog Artemis would improve the overall user experience.

gigapod, 2023-10-27:
Hi @bherbruck ,

This is a great idea and I've added it to our backlog.

### Initial Thoughts

The underlying [driver](https://github.com/sparkfun/SparkFun_ADS1015_Arduino_Library) supports float, int and unsigned int outputs, so we'll add support for these along with the *float* type in the current implementation.

This would add 16 (for a total of 24) output values, so we'll add a property to control different type modes (float, int, unsigned int), which provides an easy way to hide/show output values of different types.  But if someone wanted to, they could turn on all 24 output types....

We are currently finishing up version 1.1.0 of the firmware, with a release in mid-November. I'll have this feature added to this release. 

If you have any suggestions or don't like the proposed solution, just let us know! 

-Kirk

bherbruck, 2023-11-01:
Thanks! After looking into some arduino code, looks like I can get good values in float mode when I do something like:

```c
ADS1015 adcSensor;
adcSensor.setGain(ADS1015_CONFIG_PGA_TWOTHIRDS);
adcSensor.setSampleRate(ADS1015_CONFIG_RATE_128HZ);
float mV0 = adcSensor.getSingleEndedMillivolts(0);
```

This is the kind of behavior I would like to have in the datalogger

gigapod, 2023-11-01:
Hi @bherbruck ,

Just to verify - have you set these values via the settings menu for this device in the DataLogger?

You can set the Gain and Sample rates via the settings menu when the device is connected. 
* The property '**Sample Rate**'  has a setting option for '**128 Hz**'
* The property '**Gain**' has a setting option for '**x2/3**'

When those properties are set, they call the above methods you highlighted in the ADS1015 Arduino Library.

We're still looking into enabling switching between output data types as an option for the device, with the current type - float - being default.

-Kirk

bherbruck, 2023-11-01:
I tried all the configuration options and still couldn't get a value output above 2047mV.

Am I missing something?

gigapod, 2023-11-01:
I'm not sure and I don't have a device to test right now.

Looking at the code, does this work?
- Start Datalogger with ADS1015 attached
- Go into the settings menu and make the gain/sample changes
- Leave settings menu
- DON'T restart the DataLogger

Are the changes reflected in the output - are the values different? 

I see a possible logic error in the DL device driver for the ADS1015 - on restart, the persisted settings are not sent to the underlying Arduino library -- at least it looks that way. So - on restart, the default gain and sample rate are set. ...

Also  - I'll need to consult with the author for this driver and see if there's something else to try.

-Kirk

PaulZC, 2023-11-02:
Hi @bherbruck ,

I have been investigating this issue and I have found a bug in the 'driver' for the ADS1015. The sample rate register was being overwritten with the gain. The gain register was never updated - it was always set to 'x2' (FSR: +/- 2048mV). And due to the register bit masking, it meant the sample rate was always '0' = 128Hz.

Apologies for the inconvenience. We'll fix this in the next release.

Once you are able to, set the gain to x1 and you will be able to measure single-ended voltages up to 3.3V, and differential voltages of +/- 3.3V. The inputs cannot go above the supply voltage (3.3V), or below ground (0V). So, with that gain, you won't be able to use the full 12 bits, but at least you can use the full voltage range. Once the fix is in, the firmware will be identical to OpenLog Artemis.

I believe my colleagues will (probably) implement the unsigned and signed integer data formats you requested, since most of the code is already written. But with this issue fixed, you may not need them?

Best wishes,
Paul

bherbruck, 2023-11-14:
> you may not need them?

I think you're right.

gigapod, 2023-11-15:
The above fix outlined by @PaulZC  as well as the ability to select output types (singed, unsigned, float) is now part of [version 1.1.0](https://github.com/sparkfun/SparkFun_DataLogger/releases/tag/v01.01.00)

closed 2023-11-15 by gigapod