sparkfun/SparkFun_DataLogger #10: Same module, different addresses

MarekJar, 2023-11-03:
Greetings,

We have found out that in case we use two of the same modules on different addresses (by soldering the address bridge), for example, two buttons. The device lists two buttons connected, but outputs only data for single button.

It would be convenient to include all devices of the same type in the output and also add the possibility of outputting the I2C address of the device (at least in JSON format).

Best regards,

Marek J.

PaulZC, 2023-11-03:
Hi Marek (@MarekJar ),

I just tried two Qwiic Buttons and it seems to be working correctly:

```
Settings for: /SparkFun DataLogger IoT - 9DOF/Devices Settings

         1)  MAX17048 - MAX17048 LiPo Battery Fuel Gauge
         2)  BUTTON - The SparkFun Qwiic Button
         3)  BUTTON - The SparkFun Qwiic Button
         4)  ISM330 - ISM330 Inertial Measurement Unit
         5)  MMC5983 - MMC5983 Magnetometer

         b)  Back

Select Option:

Settings for: /SparkFun DataLogger IoT - 9DOF/Devices Settings/BUTTON

Settings:
         1)  Press Mode - Select Press Mode or Click (Toggle) Mode
         2)  LED brightness - Set the LED brightness

Outputs:
         3)  Button State - The current state of the button

         b)  Back

Select Option: 1

Settings for: /SparkFun DataLogger IoT - 9DOF/Devices Settings/BUTTON/Press Mode

Current Value of `Press Mode` =  1

Select from the following values:

         1)  Click (Toggle) Mode = 0
         2)  Press Mode = 1

         b)  Back

Select Option: 1
        [The value of Press Mode was updated to Click (Toggle) Mode = 0 ]

Settings for: /SparkFun DataLogger IoT - 9DOF/Devices Settings/BUTTON

Settings:
         1)  Press Mode - Select Press Mode or Click (Toggle) Mode
         2)  LED brightness - Set the LED brightness

Outputs:
         3)  Button State - The current state of the button

         b)  Back

Select Option: Back

Settings for: /SparkFun DataLogger IoT - 9DOF/Devices Settings

         1)  MAX17048 - MAX17048 LiPo Battery Fuel Gauge
         2)  BUTTON - The SparkFun Qwiic Button
         3)  BUTTON - The SparkFun Qwiic Button
         4)  ISM330 - ISM330 Inertial Measurement Unit
         5)  MMC5983 - MMC5983 Magnetometer

         b)  Back

Select Option: Back

Settings for: /SparkFun DataLogger IoT - 9DOF

         1)  Settings - System settings and operations
         2)  Devices Settings - Settings for connected devices

         x)  Exit

Select Option: Exit

End Settings
[I] Saving System Settings.
false,false
false,false
false,false
true,false
true,false
true,false
true,false
false,false
false,true
false,true
false,true
false,true
false,false
false,false
false,false
false,false
```

But I take your point about adding the I2C address. That would certainly help identify which button is which!

[image]

Best,
Paul

gigapod, 2023-11-03:
We'll look into this further.  

Do you see two buttons with CSV output?  

My guess is that he buttons are using the same name, so when they are added to the JSON document, the 2nd button is over writing the first button entry.

We'll see what we can do to detect this name collision with minimal overhead.

PaulZC, 2023-11-03:
Oh. Yeah. It's a JSON thing. Only one visible:

[image]

MarekJar, 2023-11-03:
Is there (going to be) a way to differentiate the two buttons based on the I2C address? Maybe include the address in the output?

gigapod, 2023-11-03:
Adding name collision logic is straight forward - when a device is loaded, check it's name across the current list of loaded devices.

We'll add behavior so that if there's a conflict, the ID is appended to the name.

So if you have a couple of buttons, the names would be:

BUTTON
BUTTON [0x6B]

This will also handle SPI devices.

We'll get this in mid November release.

MarekJar, 2023-11-09:
Thank you for your feedback.

In your example, the address is displayed only for the second/third/etc. module of the same type with different addresses.

Would it be possible to make an option in the configuration to "Always show I2C address", which will display address in the name even when only one of the type is connected?

I will give a use case scenario:

Let's say, for example, we produce analog-to-digital devices, which are based on the 4-channel ADC module (ADS1015).
We assign different addresses for the module and place each of the module to measure analog voltage for a specific device (lets call them Device 1 and Device 2).

If, for example, a user connects both modules, the system will see the address of the second one, but not the first one. This is okay, since the first one must be the one with the other address.

However, if the module connection is optional and a user connects only one module. There is no way for the control system to know if the user connected Device 1 or Device 2. This will be even worse if there are more Devices (up to 4 different addresses can be assigned to the ADS1015).

So, it would be our sincere request that there would be an option in the configuration to display I2C addresses in the name regardless of the number of modules.

Thank you and best regards,

Marek J.

gigapod, 2023-11-09:
Hi @MarekJar ,

We can do this for next weeks release.

We'll add a property, making this optional. When enabled, you'll see device names like:

```
1)  MAX17048 [x36] - MAX17048 LiPo Battery Fuel Gauge
2)  ENS160 [x53] - ScioSense ENS160 Indoor Air Quality Sensor
3)  BME280 [x77] - The Bosch BME280 Atmospheric Sensor
4)  ISM330 [p5] - ISM330 Inertial Measurement Unit
5)  MMC5983 [p27] - MMC5983 Magnetometer
```

-Kirk

gigapod, 2023-11-15:
This is fixed as outlined above - included in [release v1.1.0](https://github.com/sparkfun/SparkFun_DataLogger/releases/tag/v01.00.01)

closed 2023-11-15 by gigapod