sparkfun/SparkFun_DataLogger #8: Parameter/Character limit

MarekJar, 2023-11-03:
Hello

There seems to be some kind of parameter or character limit for the output.

We were experimenting with 6 modules on the I2C bus, but for some reason when adding modules, the last parameters are left out of the message, at least in the JSON format.

Some modules have many parameters, which causes some parameters to be left out from the output, even from the on-board sensors.
Even adding timestamp to the message causes loss of at least 1 parameter output.

Would it be possible to remove this limit or increase the number of parameters in the output to some larger number?

Best regards,

Marek J.

gigapod, 2023-11-03:
Hi @MarekJar,

The internal JSON buffer is hard coded to 1600. Obviously too conservative. 

We're making this a user accessible preference.  This will be in the next firmware update scheduled for mid November.

gigapod, 2023-11-15:
This is now a user configurable option in [version 1.1.0](https://github.com/sparkfun/SparkFun_DataLogger/releases/tag/v01.00.01)

closed 2023-11-15 by gigapod