sparkfun/SparkFun_DataLogger #11: Investigate providing alternate I2C addresses for the ACS37800 Power Meter

PaulZC, 2023-11-13:
Based on this Forum post: https://forum.sparkfun.com/viewtopic.php?p=245235#p245235

Investigate if it is possible to add alternate addresses for the ACS37800. Remembering that it will likely clash with the SCD30 and SCD40.

gigapod, 2023-11-20:
@PaulZC  What addresses do we want to add to this driver?

PaulZC, 2023-11-20:
You can change the I2C address by pulling the two IO pins to different voltage levels. Each IO pin supports four states:

[image]

I think we should go with 0x60 - 0x63, which are the possible addresses for the four possible states of one IO pin when the other is 0v:

[image]

These will overlap with the SCD30 and SCD40 but those have exact connectedConfidence.

I think attempting to implement all 16 possible states would be unwise. You can also set the address in EEPROM to _any_ valid address. It would be very very unwise to attempt to support all 112!

gigapod, 2023-11-20:
Adding these now ... will check in a few minutes ...

gigapod, 2023-11-20:
This is fixed/added to the system and will be in the next update along with an improved I2C auto load algorithm which will greatly minimize chances of address conflicts during device detection.

closed 2023-11-20 by gigapod