sparkfun/SparkFun_DataLogger #13: Add list of POSIX time zones to the docs?

PaulZC, 2023-11-20:
Maybe I missed it, but I couldn't find a link to a list of POSIX time zones in the docs.

There is a list here: https://github.com/nayarsystems/posix_tz_db/blob/master/zones.csv

I believe that list is copied and pasted from Linux /usr/share/zoneinfo/ , so maybe we could use a workflow action to auto-generate the list periodically?! ;-)

bboyho, 2023-11-28:
Hi!

I do have that linked as a note in this section => https://docs.sparkfun.com/SparkFun_DataLogger/configuration/#network-ntp-client .  Hmm, as of firmware v01.01.01, the Time Zone option was moved to this menu => https://docs.sparkfun.com/SparkFun_DataLogger/configuration/#general-time-sources . 

Hmm, I'm not sure what you mean by autogenerating the list? Should we insert that list in a section of the docs? Or in the firmware?

PaulZC, 2023-11-28:
Ah - excellent - thanks! I'd missed that...

If you copy that note into [General : Time Sources](https://docs.sparkfun.com/SparkFun_DataLogger/configuration/#general-time-sources) then I think we're done and can close this?

I spotted a typo in the NTP section: "sonze" instead of "zone".

Our future selves could write a workflow action which would generate that list of time zones from ubuntu /usr/share/zoneinfo/ , prettify it into a markdown table, copy it into the docs and push it. But only if / when we have time... ;-)

bboyho, 2023-11-29:
Ah it's ok, I forgot to move that note with it updated menu. It has been moved to the [General: Time Sources](https://docs.sparkfun.com/SparkFun_DataLogger/configuration/#general-time-sources).

Ooo, good catch. I have fixed the spelling error.

The workflow action seems nice. Might be good to keep that in mind for a future update if there is time. ;)

bboyho, 2023-11-29:
[Changues pushed ](https://github.com/sparkfun/SparkFun_DataLogger/commit/55cc85548f1165ae5cbbe8ba5b969fa2b8e3a0ae) and live. =)

closed 2023-11-29 by bboyho