sparkfun/SparkFun_DataLogger #16: Delay before probing Qwiic bus

MarekJar, 2024-01-04:
Greetings

Would it be possible to add a customizable delay on boot/startup before probing the Qwiic/I2C bus for external modules?

We would like to use the DataLogger in an application where the external modules are not powered exactly at the same time as the DataLogger, but with a slight delay (several seconds).
In our application, since the modules are powered up a little after the DataLogger, it does not register them and thus does not read the data from the modules afterwards.

Best Regards,

Marek J.

gigapod, 2024-01-04:
HI @MarekJar  -

This is a great idea and I'll add it to our next release now scheduled for late January.

As part of this release we've already added a short (2 sec) delay at startup as part of a *startup* command functionality (press a key to start without wifi, disable device autoload, don't load stored settings  and list available device drivers).  We can easily make this delay user configurable. 

-Kirk

gigapod, 2024-01-09:
Hi @MarekJar -

Is up to 60 seconds sufficient startup delay for your use case? 

-Kirk

gigapod, 2024-03-15:
Fixed: The fix for this is included in v1.2 

There is now a *startup* menu in the system. The delay of this menu can be configured in the settings/menu. 

[1.2 Pre Release](https://github.com/sparkfun/SparkFun_DataLogger/releases/tag/v01.01.99)

closed 2024-03-15 by gigapod