sparkfun/SparkFun_DataLogger #17: Support for https://www.sparkfun.com/products/14349

falkorichter, 2024-01-06:
I have an `AK9753` (https://www.sparkfun.com/products/14349) but there is no support for that device.

Is there a way I can write a "driver" for that chip myself? I was hoping the datalogger firmware was completely open source but I can only find firmware dumps :(

falkorichter, 2024-01-07:
Damn, I was hoping at least the driver part of the firmware would be open source... https://github.com/sparkfun/SparkFun_DataLogger/issues/4 confirm this is not the case and might have never been intended as such :(

falkorichter, 2024-01-08:
https://www.sparkfun.com/what_is_sparkfun

> SparkFun is a firm believer in the ideology of open source technology. 

- https://opensource.com/business/12/9/how-sparkfun-built-open-business
- https://opensource.com/article/19/8/sparkfun-creator-nathan-seidle

I hope I can somehow contribute to this great piece of software.

gigapod, 2024-01-09:
Hi @falkorichter ,

SparkFun in strongly committed to the Opens Source community, but for some products/investments we might choose not to publish the source code. This this the case for the DataLogger IoT.

We'll take a  can look at adding a driver for the AK9753 to the product for a future update.

falkorichter, 2024-01-09:
Could I offer contributions? I'm a fast learner, Android/iOS/cross platform developer since 15 years (I feel old now) and I'm more than open to sign an NDA.

falkorichter, 2024-11-17:
Hi, can I somehow offer support here?