sparkfun/SparkFun_DataLogger #19: BME68x VOC calculations on the iot logger

falkorichter, 2024-01-09:
I wanted to use my BME68x to calculate a air quality index. After some research on the breakout board, I discovered there is an embedded library by Bosch that is commonly used which uses the different sensor readings. 

That method of getting an air quality value cannot be done based on the raw data which is logged.

Would it be possible to include the Bosch library ans calculate the air quality on the iot logger? Otherwise the "gas resistance" value provided by the sensor is not very useful.

falkorichter, 2026-01-30:
any update on this?

falkorichter, 2026-02-02:
There is this pimoroni library which seems to be working all right https://shop.pimoroni.com/products/bme680-breakout

https://github.com/Zanduino/BME680

Whereas the SparkFun board does not mention this lib. https://eckstein-shop.de/SparkFun-Environmental-Sensor-Breakout-BME680-Qwiic-EN

I somewhat find the library useful, it’s simple, see https://github.com/falkorichter/fork_Adafruit_CircuitPython_SSD1305/blob/main/sensor_plugins/bme680_plugin.py

Is this sensor implementation returning the raw data which I can add to an algo as with the bme680_plugin.py

gigapod, 2026-02-03:
Thanks @falkorichter  

Looks like our system is using the Bosch provided I2C library from here:
https://github.com/boschsensortec/Bosch-BME68x-Library

Which isn't what we recommend in our hook-up guide - not sure how that occurred. 

I'll see that we switch over to https://github.com/Zanduino/BME680, which we do recommend. 

This will happen shortly after the release of v1.5 0 which we should have out on Thursday - so we have time integrate and perform some level of testing.

-Kirk

falkorichter, 2026-02-05:
I was able to implement the bme library https://github.com/falkorichter/fork_Adafruit_CircuitPython_SSD1305/blob/main/examples/mqtt_sensor_example.py by reading the values over mqtt - but there is an issue https://github.com/sparkfun/SparkFun_DataLogger/issues/38