sparkfun/SparkFun_DataLogger #20: DataLogger IoT upload format

JimKruk, 2024-01-17:
I bought this board https://www.sparkfun.com/products/20594 SparkFun DataLogger IoT with a SF TMP117 temp sensor plugged in the board. I am using the adafruit IO mqtt service. After configuring the parms for the mqtt service, wifi and the logging interval everything seems to start and connect ok:

SparkFun DataLogger IoT (c) 2023 SparkFun Electronics
Version: 01.00.04 - build 0000bb

Restoring System Settings ...restored from ESP32 Preferences
Connecting to WiFi...Connected to NETGEARMR1100
NTP startup [5 secs]...enabled
MQTT Client: connecting to MQTT endpoint io.adafruit.com:1883 ...connected
Loading devices ... 2 devices detected
TMP117 - TMP117 Precision Temperature Sensor {qwiic}
MAX17048 - MAX17048 LiPo Battery Fuel Gauge {qwiic}

Time: 2024-01-01T10:59:47
Uptime: 0 days, 00:00:09.756
External Time Source: NTP Client
Board Name: SparkFun DataLogger IoT
Board ID: SFD248966B65B7A0
[I] SD card not available
[I] WiFi - Connected SSID: NETGEARMR1100 IP Address: 192.168.1.83 Signal: Fair
[I] Battery - Not Connected
[I] System Deep Sleep: disabled
Sleep Interval: 60 seconds
Wake Interval: 120 seconds
[I] Logging Interval (ms): 60000
[I] Serial Output: Disabled
Baud Rate: 115200
[I] SD Card Output: Disabled
Rotate Period: 24 Hours

[I] IoT Services:
MQTT Client : enabled
MQTT Secure Client : disabled
HTTP IoT : disabled
AWS IoT : disabled
Azure IoT : disabled
ThingSpeak MQTT : disabled
MachineChat : disabled

[I] Connected Devices [2]:
TMP117 - TMP117 Precision Temperature Sensor {qwiic}
MAX17048 - MAX17048 LiPo Battery Fuel Gauge {qwiic}

I called the mqtt topic:
Settings for: /SparkFun DataLogger IoT/Settings/MQTT Client/MQTT Topic
Edit the value of `MQTT Topic` - data type <string>
When complete, press <Return> to accept, <ESC> to discard
MQTT Topic = zzzzzzzz/feeds/datalogger.tmp117

I disabled the outputs for the 3 fields of the max17048 device in /SparkFun DataLogger IoT/Devices Settings/MAX17048

Settings for: /SparkFun DataLogger IoT/Devices Settings/MAX17048
Outputs:
1) Voltage (V) - Battery voltage (Volts)
2) State Of Charge (%) - Battery state of charge (%)
3) Change Rate (%/hr) - Battery charge change rate (%/hr)

When I open the web console of adafruit IO I am seeing updates once a minute but in the 'value' of the tmp117 feed I see this as the value:

{"TMP117":{"Temperature (C)":27.96875},"MAX17048":{}}

I just want to see the data value for "Temperature (C)" of 27.96875 not the entire string above

Is there a config setting for that somewhere ?
thanks

gigapod, 2024-01-17:
Hi @JimKruk 

Note: I moved this issue to our Firmware repository for the DataLogger devices.

The generic MQTT connection outputs a JSON object that groups readings in sub-objects that are named based on the connected devices. This is what your'e seeing above. 

Adafruit IO has a specific JSON format specified for their MQTT JSON payload - it's outlined here: https://io.adafruit.com/api/docs/mqtt.html#adafruit-io-mqtt-api

This isn't surprising, each IoT service defines their own expected payload. 

To make this work correctly, we would need to build an Adafruit IO driver. 

I'll add it to our backlog for evaluation. 

-Kirk

JimKruk, 2024-01-17:
Thank you for the update

gigapod, 2026-01-21:
This feature will not be implemented

closed 2026-01-21 by gigapod