sparkfun/SparkFun_DataLogger #21: Feature Request - output logs over serial/USB

stcase, 2024-01-24:
Hi,
Would it be possible to add support for accessing saved logs from the SD card over the serial connection (so the device doesn't need to be turned off and the SD card manually removed)? It would be nice for cases where the logger loses access to a serial connection or wifi.

Essentially, it could be the same thing as what was done for the Artemis logger here: https://github.com/sparkfun/OpenLog_Artemis/issues/33

gigapod, 2024-01-24:
Hi @stcase 

One solution for this is available with the next release (later this month) -  allows you to browse and download log files via a web page the datalogger serves. 

We'll add a serial option to our development  backlog. 

-Kirk

stcase, 2024-01-25:
Awesome, thanks!

gigapod, 2024-03-15:
The web download ablity is in v1.2  I'm leaving tihis issue open - just in case we want to look into the serial download option...

[1.2 Pre Release](https://github.com/sparkfun/SparkFun_DataLogger/releases/tag/v01.01.99)