sparkfun/SparkFun_DataLogger #23: Feature Request: Wait for GNSS datetime validity before syncing ESP32 RTC with GNSS

adamgarbo, 2024-03-08:
Howdy,

I would like to suggest the if the Datalogger IoT is configured to use a GNSS as its time sourece, that it waits until it has a valid time from the GNSS before synchronizing its RTC from the GNSS and begins to log data 

When connected to a SparkFun u-blox SAM-M8Q board, I have observed that when there is no GNSS fix, the default datetime is 1970-12-31T17:00:00. As the GNSS transitions from no fix, to 2D to 3D, `General.Time` is never updated (see below). A system reset is required for the Datalogger to start using the GNSS as the time source. I am familiar with the types of checks that can be programmed with respect to u-blox datetime validity flags, and I suspect these are only checked at start-up.

Cheers,
Adam

```
General.Time,GNSS.SIV,GNSS.Fix Type,GNSS.Ground Speed (m/s),GNSS.PDOP,GNSS.TOW (ms),GNSS.ISO8601,GNSS.Fix Type (String),GNSS.Carrier Solution (String)
1970-12-31 17:00:04,0,0,0,99.99,4000,2015-10-18T00:00:04Z,none,none
1970-12-31 17:00:05,0,0,0,99.99,5000,2015-10-18T00:00:05Z,none,none
...
1970-12-31 17:00:23,0,0,0,99.99,483778000,2015-10-23T14:22:41Z,none,none
1970-12-31 17:00:24,0,0,0,99.99,483779000,2015-10-23T14:22:42Z,none,none
1970-12-31 17:00:25,0,0,0,99.99,483780000,2015-10-23T14:22:43Z,none,none
1970-12-31 17:00:26,0,0,0,99.99,483781000,2015-10-23T14:22:44Z,none,none
1970-12-31 17:00:27,0,0,0,99.99,483782000,2024-03-08T14:22:45Z,none,none <- Likely valid datetime flags
1970-12-31 17:00:28,0,0,0,99.99,483783000,2024-03-08T14:22:46Z,none,none
1970-12-31 17:00:29,0,0,0,99.99,483784000,2024-03-08T14:22:47Z,none,none
1970-12-31 17:00:30,0,0,0,99.99,483785000,2024-03-08T14:22:48Z,none,none
1970-12-31 17:00:31,0,0,0,99.99,483786000,2024-03-08T14:22:49Z,none,none
1970-12-31 17:00:32,0,0,0,99.99,483787000,2024-03-08T14:22:50Z,none,none
1970-12-31 17:00:33,0,0,0,99.99,483788000,2024-03-08T14:22:51Z,none,none
1970-12-31 17:00:34,0,0,0,99.99,483789000,2024-03-08T14:22:52Z,none,none
1970-12-31 17:00:35,0,0,0,99.99,483790000,2024-03-08T14:22:53Z,none,none
1970-12-31 17:00:36,0,0,0,99.99,483791000,2024-03-08T14:22:54Z,none,none
1970-12-31 17:00:37,0,0,0,99.99,483792000,2024-03-08T14:22:55Z,none,none
1970-12-31 17:00:38,0,0,0,99.99,483793000,2024-03-08T14:22:56Z,none,none
1970-12-31 17:00:39,0,0,0,99.99,483794000,2024-03-08T14:22:57Z,none,none
1970-12-31 17:00:40,0,0,0,99.99,483795000,2024-03-08T14:22:58Z,none,none
1970-12-31 17:00:41,0,0,0,99.99,483796000,2024-03-08T14:22:59Z,none,none
1970-12-31 17:00:42,0,0,0,99.99,483797000,2024-03-08T14:23:00Z,none,none
1970-12-31 17:00:43,3,2,0.032,2.93,483797998,2024-03-08T14:23:01Z,2D,none <- 2D fix/valid datetime flags
1970-12-31 17:00:44,3,2,0.009,2.93,483799000,2024-03-08T14:23:02Z,2D,none
1970-12-31 17:00:45,3,2,0.079,2.93,483800000,2024-03-08T14:23:03Z,2D,none
1970-12-31 17:00:46,3,2,0.076,2.93,483801000,2024-03-08T14:23:04Z,2D,none
1970-12-31 17:00:47,3,2,0.029,2.93,483802000,2024-03-08T14:23:05Z,2D,none
1970-12-31 17:00:48,3,2,0.08,2.93,483803000,2024-03-08T14:23:06Z,2D,none
1970-12-31 17:00:49,3,2,0.105,2.93,483804000,2024-03-08T14:23:07Z,2D,none
1970-12-31 17:00:50,3,2,0.123,2.93,483805000,2024-03-08T14:23:08Z,2D,none
1970-12-31 17:00:51,3,2,0.076,2.93,483806000,2024-03-08T14:23:09Z,2D,none
1970-12-31 17:00:52,3,2,0.055,2.93,483807000,2024-03-08T14:23:10Z,2D,none
1970-12-31 17:00:53,3,2,0.013,2.93,483808000,2024-03-08T14:23:11Z,2D,none
1970-12-31 17:00:54,4,3,0.293,22.08,483809000,2024-03-08T14:23:11Z,3D,none <- 3D fix/valid datetime flags
1970-12-31 17:00:55,4,3,0.189,22.06,483810000,2024-03-08T14:23:12Z,3D,none
1970-12-31 17:00:56,4,3,0.152,22.05,483811000,2024-03-08T14:23:13Z,3D,none
1970-12-31 17:00:57,4,3,0.06,22.04,483812000,2024-03-08T14:23:14Z,3D,none
1970-12-31 17:00:58,4,3,0.069,22.02,483813000,2024-03-08T14:23:15Z,3D,none
1970-12-31 17:00:59,4,3,0.08,22.01,483814000,2024-03-08T14:23:16Z,3D,none
1970-12-31 17:01:00,4,3,0.045,21.99,483815000,2024-03-08T14:23:17Z,3D,none
1970-12-31 17:01:01,4,3,0.052,21.98,483816000,2024-03-08T14:23:18Z,3D,none
1970-12-31 17:01:02,4,3,0.07,21.97,483817000,2024-03-08T14:23:19Z,3D,none
```

gigapod, 2024-03-08:
Hi @adamgarbo -

What happens now:
- There is a "On initialize" phase that is called on the clock system during startup. This is called after the devices are loaded, initialized and prefs restore if avilable (on board prefs system, or from a JSON file)
- The clock system checks the "reference" clock during this "initalization" call. If the source epoch is "valid", the system time is set, otherwise it is not.
- The reference is then checked again at the value of the "Update Interval" property setting - The default is 60 minutes, but this is adjustable. 

So what you're seeing is what would be expected given this logic.

I belieave we could modify this logic to something like this:

If the system clock hasn't been intalized from the external ref source and  the ref clock isn't valid
- Set the update period/timer to 1 minute
- Once the ref clock is valid, and system time set, reset the timer is set to the user set prefernce. 

This looks easy enough to do ... 

We have a preview of firmware version 1.2 going out on Monday - let me see if I can sneek this update into this release 

-Kirk

adamgarbo, 2024-03-08:
Thanks @gigapod, for the quick reply!

I appreciate the insights into how things currently work with respect to time sources. In my use case, I opted to rely solely on the GNSS for timing as opposed to an RTC. The main question/concern for me was "How do I know that my data is being logged with the correct timestamp?". Of course, I could have always ensured the GNSS timestamps were being logged separately and ignore General.Time.

Looking forward to the next firmware release!

Cheers,
Adam

gigapod, 2024-03-11:
Hi @adamgarbo -

Logic was added so if the reference clock isn't *valid* at startup/initialization - the revisit / synch period is set to 1 minute. The reference clock is checked every minute.

Once the ref clock is valid and system time is set to the ref clock, the clock synch period is reverted to the preference setting. 

This is in the 1.2 Preview - which is now pushed to tomorrow or Wednesday .. but will be out this week. 

-Kirk

gigapod, 2024-03-15:
Fixed: The fix for this is included in v1.2 

[1.2 Pre Release](https://github.com/sparkfun/SparkFun_DataLogger/releases/tag/v01.01.99)

closed 2024-03-15 by gigapod