sparkfun/SparkFun_DataLogger #25: configuration file export / import - https://www.sparkfun.com/products/20594

mvschaefer, 2024-03-23:
i have been using various serial terminals to set various configuration settings and it is an abysmal experience.

it seems like having an export / import to / from a csv / json file that has all available options would provide a much better user experience.

gigapod, 2024-03-23:
Hi @mvschaefer ,

You can save/restore the configuration settings from a json file the SD Card. If no settings are set in the system on startup, and a file called `datalogger.json` exists on the SD card, settings will be loaded from this file.

You can create this file using the Serial menu system - under the Save Settings menu > "Save to Fallback" option. Note: This won't output any "secret" properties (passwords, keys ...etc) - just empty values for these entries.  Outlined [here](https://docs.sparkfun.com/SparkFun_DataLogger/configuration/#general-save-settings)

We use this capablity extensivly for our development and testing. 

And with the latest firmware - in *preview* right now, you can stream in the JSON settings option via the serial console if you want. See the command [!json-settings ](https://github.com/sparkfun/SparkFun_DataLogger/blob/main/docs/relnotes/rn_v010200.md#quick--commands)

-Kirk

gigapod, 2024-03-23:
Also - if there are any details you have as to why the serial console isn't working correct - issues seen, terminal software used ..etc - let us know and we can test it on our end.

closed 2024-03-23 by gigapod

mvschaefer, 2024-03-24:
@gigapod  - thanks for the response!  i suspect with the json file capabilities, my issues with the menu system will be resolved.