sparkfun/SparkFun_DataLogger #28: Using Flash Download Tool of Espressif to update the firmware. (reopen of issue#3)

mvschaefer, 2024-03-25:
ive searched the sparkfun repo(s) and can't find esptool.py or instructions for re-flashing .bin files to the 9dof IoT board.

gigapod, 2024-03-26:
The command needed are in the notes of this release: https://github.com/sparkfun/SparkFun_DataLogger/releases/tag/v01.00.01

esptool.py can be installed using pip: https://docs.espressif.com/projects/esptool/en/latest/esp32/installation.html

Once the board is reflashed, you can use the built in update tools to upgrade to the latest firmware version.

closed 2024-03-26 by gigapod

mvschaefer, 2024-03-26:
esptool --chip esp32 --port COM4 --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 16MB 0x1000 "./v01.00.01.bootloader.bin" 0x8000 "./v01.00.01.partitions.bin" 0x20000 "./v01.00.01.bin"
esptool.py v4.7.0
Serial port COM4
Connecting....
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: a0:b7:65:5e:4e:90
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00020000 to 0x0024dfff...
Compressed 18960 bytes to 13074...
Wrote 18960 bytes (13074 compressed) at 0x00001000 in 0.6 seconds (effective 273.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 143...
Wrote 3072 bytes (143 compressed) at 0x00008000 in 0.0 seconds (effective 507.2 kbit/s)...
Hash of data verified.
Compressed 2284944 bytes to 1170638...
Wrote 2284944 bytes (1170638 compressed) at 0x00020000 in 31.4 seconds (effective 581.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

mvschaefer, 2024-03-26:
NOTE: the rename (local copy/paste) of the files to: v01.00.01.*.bin

NOTE: esptool.exe called from a local .bat file w/the appropriate COM port

mvschaefer, 2024-03-26:
after flashing completed, unplug from USB-C, plug back in:

- Version: 01.01.01  - build 0000f4