sparkfun/SparkFun_DataLogger #30: UART access?

cuzzard, 2024-04-20:
I would like to try this board to log a UART sensor. When is UART access expected to be included?
Thanks

gigapod, 2024-04-22:
Easy enough to add to a near term release

Would you expect something like this?
- external UART device connected to the TXD/RXD pins of the DataLogger IoT board
- UART datalogging enabled in DataLogger prefs
- Every log cycle the datalogger will read lines of data from the datalogger
        - Dumping each line to the output stream as single lines
        - Dumping everything it reads into the output streams as a single blob of text

And for the data read from the UART, how would you expect to see it in the log stream - just the value of a field in the CSV or JSON results? 

Any input helpful!

cuzzard, 2024-04-23:
Yes, I think what you have outlined would be suitable.

I am assuming the timestamp would also be able to be included as per the current options, so you could have Timestamp and then the data string  as one line per data pulse.

I have submitted another item for PPS timestamps; this would make things even more accurate for sensors collecting data at rates upwards of 100 Hz.

Also, having the read accept the native output string of whatever device/sensor a user decides to connect would be most useful I think. Then users could format their data after, if required.

gigapod, 2024-05-02:
Thanks for the follow-on information. I've added this to our backlog for the next release. 

The current thought is this would just be another output data value that is managed by the Logger object in the system -  so everything that's in the current Logging options in the system applies (timestamp, sample number ...etc). And the data is sent to whatever the output is (serial, file, AWS, Azure ...etc). 

We do something similar to this in the Artemis OpenLog firmware (which is a different codebase), so we can reference that functionality/implementation. 

Timeline - end of quarter (June) is what we're looking at for a next release - will post a alpha and beta before this.

Thanks
-Kirk