sparkfun/SparkFun_DataLogger #31: PPS time pulse ref clock

cuzzard, 2024-04-21:
Is there any future plan to implement an interrupt using the PPS from a GNSS? This would help keep the most accurate reference clock, rather than relying on NMEA strings over Qwiic which can be delayed by 100+ ms.

gigapod, 2024-05-02:
This is added to our next release backlog. 

The concept we have is that if a PPS signal is connected to a GPIO pin on the DataLogger,  the detection of the signal would trigger a log event. This event is in addition to any timer set - (if the timer is set).

cuzzard, 2024-05-03:
Sounds practicable, Thanks.

Am I correct in assuming the system UTC timestamp in the log file will still be 'old' i.e. a PPS logged event could obtain a timestamp of 09:12:57.150 if the NMEA timestamp was 150 ms behind the PPS?

Also, if it could be possible to distinguish PPS log events from other events, this can ensure some sort of post-processing step could be implemented and also samples could then be accurately positioned in time with other sensors e.g. from isolated systems also matched to GNSS PPS?

This could be something as simple a '-PPS' flag being tagged to the sample.

gigapod, 2024-05-03:
The timestamp the logging system uses is getting the time from the system (esp32 in this case).   

The GNSS device is logged also (if enabled) and outputs the time/date values (again if enabled).  This are via the getHour(), getMinute(), getSecond() methods of the u-blox Arduino library.

We can look at tagging the trigger for the event in the output stream - right now a non-named event/signal is used to cause a log action (timer, button ...etc). 

-K

cuzzard, 2024-05-03:
Understood.

It would be valuable to be able to distinguish the difference/offset (if any) between the PPS event/signal and the getSecond() output timestamp of the U-Blox Arduino library.

Therefore, having the PPS event/signal tagged with some delimiter/name would be my suggestion for those seeking the highest accuracy in time to use.

Thanks.

Rob

gigapod, 2025-07-02:
... Trigger/event source is also now available in the output/log stream.  There is an option to enable/disable this in the logger settings menu. 

Values include: "PPS", "SERIAL" (if logging input from a connected serial device), "Timer", and a user provided name for an Interrupt driven trigger.