sparkfun/SparkFun_DataLogger #33: Cannot change Logging Timer Interval

datenen, 2024-05-23:
To adjust the frequency of logging events you are supposed to go into menu item 7, option 1, but there does not seem to be the ability to edit the default value of 15000. If you try to type in another number it is either multiplied by the 15000 (type in 100 and get 1500000) or appended to the 15000 (type in .066 and you get 15000066).  I have tried everything I can think of to get the timer interval to 1000 (once persecond) with no success, sure would appreciate getting this fixed for my application! Thanks! See attachment: the [D comes from trying to input a backspace from the keyboard to change the default value.
[image]

gigapod, 2024-05-23:
You need a serial terminal client that can support interactive edits - most if not all modern solutions do though. What serial terminal client are you using and what platform are you on? 

We've tested using Terra on Windows, on Linux/Unix/MacOS the old command line program 'minicom' and 'screen' work well.

datenen, 2024-05-23:
Thanks Kirk! I had used Coolterm on both platforms (PC/Mac) with no luck. Per your email downloaded TeraTerm 4.107 for Windows, and it appears to have let me change the setting.

Appreciate it! you might want to reflect this in your online doc for the Datalogger iot.

Best,

David

gigapod, 2024-05-23:
Great to hear - yes, I'll see that we add a note in our documentation about the need for a serial termal client that supports edit characters.

datenen, 2024-05-23:
Cool! FYI I confirmed minicom works fine on the Mac as well. Not for nothing, but if I had read "look for a serial terminal client that supports edit characters" I would have no idea what to try. You suggesting specific packages like Tera Term or minicom did the trick for me!

Thanks again!!

David

gigapod, 2024-05-23:
@bboyho - can you add a bit in our docs for the DataLogger IoT about needed a serial client that supports edit characters (see above). I know Terra Term, minicom and screen work.

bboyho, 2024-07-30:
Ahh! I'm sorry, I just saw this issue. Updates have been added to the tutorial.

closed 2024-07-30 by bboyho